sunnypilot/sunnypilot #42: Add Toggle for enabling Tethering on boot

khoi, 2022-02-20:
People outside US are likely to have an unlimited/unthrottled data plan, and using Comma device for sharing network would be great

wallneradam, 2022-03-08:
I've created a service (systemd), which enables hotspot, when the car has lost WIFI connection, then disables it when the car is near home (by GPS).

If you interested I can share the code. It is not integrated into Openpilot UI, it is a python service, with a systemd unit file. And the home (and/or work) coordinates can be set with a constants. Though it can be integrated, just it was good for me.

khoi, 2022-07-04:
do share it please @wallneradam

sunnyhaibin, 2022-07-08:
This will be added in sunnypilot 0.8.14-2. Thanks for the suggestion!

closed 2022-07-08 by sunnyhaibin