sunnypilot/sunnypilot #47: ui process cpu usage too high.    slow frame rate

ajouatom, 2022-03-09:
Thanks for your development,
I ran into the problem of "slow frame rate" while referencing your code.
The cause of this problem was none other than the ButtonsWindow class problem.

It was solved by modifying it as follows

void ButtonsWindow::updateState(const UIState &s) {
    if (m_dynamic_lane_profile != s.scene.dynamic_lane_profile) {
        m_dynamic_lane_profile = s.scene.dynamic_lane_profile;
        if (s.scene.dynamic_lane_profile == 0) {
            dlpBtn->setStyleSheet(QString("font-size: 45px; border-radius: 100px; border-color: %1").arg(dlpBtnColors.at(0)));
            dlpBtn->setText("Lane\nonly");
        }
        else if (s.scene.dynamic_lane_profile == 1) {
            dlpBtn->setStyleSheet(QString("font-size: 45px; border-radius: 100px; border-color: %1").arg(dlpBtnColors.at(1)));
            dlpBtn->setText("Lane\nless");
        }
        else if (s.scene.dynamic_lane_profile == 2) {
            dlpBtn->setStyleSheet(QString("font-size: 45px; border-radius: 100px; border-color: %1").arg(dlpBtnColors.at(2)));
            dlpBtn->setText("Auto\nLane");
        }
    }
 }

sunnyhaibin, 2022-05-07:
@ajouatom Thank you for catching this! This will be added in 0.8.12-5 along with the other new features 😊 Will close this issue when 0.8.12-5 is out.

sunnyhaibin, 2022-07-06:
Resolved in [sunnypilot 0.8.14](https://github.com/sunnyhaibin/sunnypilot/tree/0.8.14-prod-c3).