sunnypilot/sunnypilot #58: Honda Pilot 2021: ACC does not always engage

pacman2790, 2022-04-27:
Is anyone else having this issue? Sometimes when engages ACC the gas pedal doesn’t engage. I also have a comma pedal.

wumbology247, 2022-05-01:
Same issue with my 2018 Accord. I hit set then about 15 seconds later ACC cancels even though autosteer continues to work. Pretty annoying because I love the fork otherwise.

sunnyhaibin, 2022-05-01:
@pacman2790 @wumbology247 thanks for the report. I have pushed a fix (https://github.com/sunnyhaibin/sunnypilot/commit/9d773919bf44d9854531f83881ae9292efa9cdd7) to `0.8.12-prod-full-honda`, see if it helps with the issue?

pacman2790, 2022-05-07:
@sunnyhaibin thanks for taking a look! I gave that a try and still have the issue. I forgot to mention in the original that I have a comma pedal.

sunnyhaibin, 2022-05-07:
> @sunnyhaibin thanks for taking a look! I gave that a try and still have the issue. I forgot to mention in the original that I have a comma pedal.

Can you share your device with me (https://connect.comma.ai/, gear icon, share by email, share icon again to confirm), set the device name to your year/make/model so I can find it in the list, and DM me the dongle ID, and I'll take a look.

pacman2790, 2022-05-07:
@sunnyhaibin I've reinstalled and reflashed and will let you know if this works. When I try to share by email I use _sunnyhaibin_ however it says it could not find the user.

pacman2790, 2022-05-09:
Just to follow-up, it does not appear that installing and reflashing the panda worked. I also tried sharing via email again and still received "could not find user"

pacman2790, 2022-05-14:
@wumbology247 were you able to get this resolved with the update?

wumbology247, 2022-08-12:
> @wumbology247 were you able to get this resolved with the update?

Sorry for the delay there but the issue seems to have resolved. thank you! 
Also I was wondering if the distance profiles are supported on the 2018 Accord. I live in LA and a closer following distance would come in handy :)

sunnyhaibin, 2022-12-05:
Is this still an issue? I have some fixes in the latest [`test-c3`](https://github.com/sunnyhaibin/sunnypilot/tree/test-c3) branch. If this is still an ongoing issue, please feel free to open another issue and reference this one.

Thanks!