sunnypilot/sunnypilot #87: Kia EV6 getting CAN error on test-c3 branch

alenl2, 2022-11-29:
When switching back to test-c3-apollo branch it works fine but something brakes with test-c3.

Error logs show "0x1FA NOT SEEN" from opendbc/can/parser.cc

alenl2, 2022-11-30:
It has something to do with reading the speed limit.

Commenting out these 2 lines fixes the parser

https://github.com/sunnyhaibin/sunnypilot/blob/f34deb644419752d8ba7d50f794901c0341674d7/selfdrive/car/hyundai/carstate.py#L857
https://github.com/sunnyhaibin/sunnypilot/blob/f34deb644419752d8ba7d50f794901c0341674d7/selfdrive/car/hyundai/carstate.py#L858

sunnyhaibin, 2022-12-05:
@alenl2 Does your EV6 have HDA1, HDA2, or no HDA? Can you share a route from your car?

> • Share your device with me (https://connect.comma.ai/, gear icon, share by email, sunnyhaibin@gmail.com, share icon again to confirm), set the device name to your year/make/model and your Discord username (with #xxxx) so I can find it in the list.

Feel free to reach out to me in the sunnypilot discord server's #test-feedback channel to continue.

alenl2, 2022-12-08:
It's HDA2 car but looks like the latest branch (test-c3-20221207) fixe the issue.

closed 2022-12-08 by alenl2