sunnypilot/sunnypilot #90: Toyota Corolla 2019

autorun1992, 2022-12-04:
Manager failed to start

Traceback(most recent call last):
File"<frozen importlib._bootstrap >,line 219, in

sunnyhaibin, 2022-12-05:
Try to [factory reset](https://commasupport.zendesk.com/hc/en-us/articles/4413447038103-Performing-a-Factory-Reset#:~:text=Reset%20the%20comma%20three%20by,reset%20screen%20will%20come%20up.) and reinstall with the same URL. A reinstall usually fixes this issue. Feel free to reopen the issue if you continue to see this error.

Thanks!

closed 2022-12-05 by sunnyhaibin

Sliicy, 2025-03-23:
For anyone in the future that needs help, the link mentioned above is dead so just head to [flash.comma.ai](https://flash.comma.ai) and follow the steps there to reflash your device and factory reset it.