sunnypilot/sunnypilot #109: 2023 Toyota Highlander not in SP Vehicles List

OleMissGuy, 2023-02-08:
I installed using installer.comma.ai/sunnyhaibin/test-c3 (sunny pilot 2022.11.13/test-c3/f34deb6/Nov 13) and got a car not supported error.  It directed me to the SP Vehicles menu to manually select my car (2023 Toyota Highlander).  The SP Vehicles list only lists Highlanders up to model year 2022.  I have a comma3 (dongle ID 9EF0 941B E9F8 057D).  It works fine on a 2017 Highlander (which I traded in for the 2023 Highlander).
I tried the standard openpilot release at openpilot.comma.ai, but also get the same unsupported car error.  I ordered a new harness, which looks identical to the old one.  The same problem occurs for either harness.  I made a note in the discussion section of openpilot, but haven't seen any replies for several days.
Any ideas you may have would be greatly appreciated.  I look forward to being able to use my comma3 again.  I will be glad  to send any additional info/pictures  you need.

Thanks.

OleMissGuy, 2023-02-08:
I thought the following images may help speed up finding issue:

First, this is the version:
[image]

This is the dongle ID:
[image]

This shows the base SP Vehicles menu.  At the top you can manually select car:
[image]

This shows the Toyota Highlander section after scrolling down the list.  Notice the 2023 Highlander is not listed.  In fact, I saw no 2023 Toyota model  listed:
[image]

I purchased this comma3 last summer and installed in a 2017 Highlander.  It worked fine.  Could it be I have to recompile agnos to see the 2023 models?

Thanks.

sunnyhaibin, 2023-03-25:
2023 Toyota Highlander is now supported in the latest sunnypilot.

closed 2023-03-25 by sunnyhaibin