sunnypilot/sunnypilot #127: Gap Adjust Cruise can't switch from steering wheel.

one0410, 2023-03-14:
2022 Hundai ioniq 5 (soundeast asia model)
In Gap Adjust Cruise setting, I can't switch gap mode from steering wheel but only on touch screen UI.

sunnyhaibin, 2023-03-25:
- What settings did you have for GAC under "SP - Controls"?
- Was Experimental Mode enabled, or the Experimental icon stay on the top right of the driving screen?

one0410, 2023-03-28:
[image]
and I found that no matter what range I switched from UI.
The range indicator didn't change on the dashboard, it always showed 4 level.

sunnyhaibin, 2023-03-28:
> [image] and I found that no matter what range I switched from UI. The range indicator didn't change on the dashboard, it always showed 4 level.

If you are on the latest branch, it is because you have "Experimental Mode" set to ON. GAC has zero effect when in Experimental Mode.

one0410, 2023-03-28:
OK, thanks for your explanation. I'll keep trying.

closed 2023-03-28 by one0410