sunnypilot/sunnypilot #133: OpenStreetMap Uploads

tomhughes, 2023-03-22:
I'm afraid that I have had to revoke the OAuth token you are using to upload traces to OpenStreetMap. There are several reasons for this:

* Firstly you have committed a bearer token which gives access to the API under your account to this repository which is not acceptable - bearer tokens should be treated as secret as they given access to operate (in limited ways) as you.
* Secondly you are uploading vast numbers of traces under one account that is using an email address that cannot in any way handle the number of upload notifications it is getting which is causing yahoo to block our email server which is impacting other users and causing thousands of emails at a time to get stuck in our queue.

Please do not attempt to reauthorise your application using a new token without first addressing these issues.

tomhughes, 2023-03-22:
It would also be good to set a user agent in place of the default python-requests one.

sunnyhaibin, 2023-03-23:
> It would also be good to set a user agent in place of the default python-requests one.

Thanks for the suggestion. The bearer token will no longer be in the source code and the user-agent is set for additional obscurity.
- https://github.com/sunnyhaibin/sunnypilot/pull/135

closed 2023-03-26 by sunnyhaibin