sunnypilot/sunnypilot #143: LTA unavailable warning sound after unplug power from comma

Shawnfoto, 2023-03-27:
Version: 8.14
Model: Prius Prime 22"

If I turn off my car and unplug USB-C from comma, it will show LTA unavailable warning sound.
But if turn off car for a while, then, it won't show and alert it.

dbzx6r, 2023-05-26:
I believe this is by design, you can just ignore it. This also seems like something for the main OpenPilot github: https://github.com/commaai/openpilot

sunnyhaibin, 2023-05-28:
Try installing the latest `dev` branches and see if this issue can be replicated. Per @dbzx6r 's suggestion, see if this issue is also replicable with stock openpilot installed. Feel free to comment to reopen the issue if the issue persists with the latest branches.

closed 2023-05-28 by sunnyhaibin