sunnypilot/sunnypilot #151: Comma 3 connect offline

polytech90, 2023-04-18:
Hello Folks, 
I have my comma 3 just for 3 days and after setting up my device and installing the stock version of open pilot 0.9.1 and using my smartphone hotspot successfully the comma 3 connected and showed "CONNECTED ONLINE "
But after that it shows that is connected to my hotspot in the network settings but it shows " CONNECT OFFLINE " in the main screen. 
However at home using an external power and connected to my home wifi.
the main screen shows CONNECT ONLINE at all times. 
any help please to this issue would be greatly appreciated. 
Thanks.

dbzx6r, 2023-05-26:
I think you meant to put this in the official OpenPilot repo: https://github.com/commaai/openpilot

sunnyhaibin, 2023-05-28:
Per @dbzx6r 's suggestion, please reopen the issue on [commaai's openpilot GitHub repository](https://github.com/commaai/openpilot) as it is not sunnypilot.

closed 2023-05-28 by sunnyhaibin