sunnypilot/sunnypilot #184: mazda cx5 2023 longitudinal control is not working

lijianw2, 2023-06-13:
### Describe the bug

Custom ACC can be toggled but it doesn't seem to change the behavior of stock ACC on Mazda CX5 2023. 
Also, experimental mode cannot be toggled. 

### Which car does this affect?

Mazda CX-5 2023

### Provide a route where the issue occurs

n/a

### openpilot version

0.9.2

sunnyhaibin, 2023-06-16:
Does this car auto-resume with stock openpilot? Experimental Mode is not supported with Custom Stock Longitudinal Control at the moment.

sunnyhaibin, 2023-06-28:
Please feel free to reopen the issue or reach out to us directly in the [sunnypilot discord](https://discord.sunnypilot.com) if this issue persists in the latest `release-c3` branch.

closed 2023-06-28 by sunnyhaibin