sunnypilot/sunnypilot #185: Ford/Tesla missing car controller "c" when they call `create_common_events()`

twilsonco, 2023-06-17:
### Describe the bug

A car control objected is expected in the `create_common_events()` function but missing from the call for Ford and Tesla.
[image]

Here's the thrown exception on a ford maverick
[image]

### Provide a route where the issue occurs

foo

### openpilot version

SP 0.9.2.2

sunnyhaibin, 2023-06-17:
Thanks for the report. Ford is being tracked and worked on at the moment; Tesla is currently not supported and should be behind dashcam mode upstream. I will send out an update on the sunnypilot Discord server when it is ready for Ford users to test.

sunnyhaibin, 2023-08-10:
This issue has been resolved in the latest `dev-c3`, and will make its way to `staging-c3` and `release-c3`.

closed 2023-08-10 by sunnyhaibin