sunnypilot/sunnypilot #187: Toyota/Lexus - Enhanced Speed Control - ACC speed setting

markhatchell, 2023-06-28:
Hello, Thanks for a great version of openpilot with wonderful features!

I was curious if there were plans to expand the Enhanced Speed Control to Toyota/Lexus models.

I can help test if needed.

Car Make: Lexus
Car Model: NX300
Car Year: 2018
Hardware: Comma Three

Thanks,
Mark

sunnyhaibin, 2023-06-28:
Enhanced Speed Control works on all Toyota/Lexus platforms that use openpilot longitudinal control. It seems that your car by default uses stock Toyota longitudinal control. According to [CARS.md](https://github.com/commaai/openpilot/blob/master/docs/CARS.md), You will need to install the [smartDSU](https://github.com/commaai/openpilot/wiki/Smart-DSU) to enable openpilot longitudinal control on your car.

[image]
[image]

closed 2023-06-28 by sunnyhaibin

markhatchell, 2023-06-28:
Thanks for the reply. I had found the smartDSU hardware as being a requirement for actual longitudinal control however, I was under the impression that the enhanced speed control feature sent messages to the cars stock ACC to adjust the speed setting letting the stock system still do longitudinal control. That is what I was asking about getting working.

sunnyhaibin, 2023-06-28:
> Thanks for the reply. I had found the smartDSU hardware as being a requirement for actual longitudinal control however, I was under the impression that the enhanced speed control feature sent [...]

Enhanced Speed Control requires openpilot longitudinal control or Custom Stock Longitudinal Control. Toyota/Lexus currently is not compatible with Custom Stock Longitudinal Control. For this platform, the best option is to install smartDSU and to allow openpilot longitudinal control to use Enhanced Speed Control.