sunnypilot/sunnypilot #191: ev6 custom stock longitudinal control can't run

kyy3067, 2023-07-09:
### Describe the bug

ev6 2022 with hda2 
custom stock longitudinal control can't run
[image]

### Which car does this affect?

ev6 2022 with hda2

### Provide a route where the issue occurs

staging-c3

### openpilot version

staging-c3

### Additional info

[image]
i would like to use it

sunnyhaibin, 2023-07-10:
Custom Stock Longitudinal Control is temporarily disabled for HKG CAN-FD cars since sunnypilot 0.9.2.1 due to an upstream openpilot issue with the Red Panda firmware. comma mentioned that the fix is targeted to be implemented in 0.9.4 but no ETA yet. If you would like to use the feature without issues, you can install the `test-c3-vw-custom-stock-long` branch for the time being (it is still on 0.9.2).

You can follow the thread in the sunnypilot discord under the `#dev-feedback` channel or the `#hyundai-kia-genesis` channel for updates on this feature being reimplemented.

`DISABLED: Hyundai/Kia/Genesis CAN-FD platforms `

https://github.com/sunnyhaibin/sunnypilot/blob/7ec767f53fbc18fa6b45f72dc478f24196a124c6/CHANGELOGS.md?plain=1#L45-L46
https://github.com/sunnyhaibin/sunnypilot/blob/7ec767f53fbc18fa6b45f72dc478f24196a124c6/CHANGELOGS.md?plain=1#L67-L71

closed 2023-07-10 by sunnyhaibin

kyy3067, 2023-07-10:
thanks you 
you veryvery nice