sunnypilot/sunnypilot #197: ev6 stop&go bug

kyy3067, 2023-08-07:
### Describe the bug

Hello.
I'm using the Sunny Pilot well.

my car is kia ev6 with hda2 / 2022

However, if you restart after stopping, there will be a bug.

"Can Bus Disconnected:Likely Faluty Cable"

I thought it was a cable problem, so I replaced all the cables
Occasionally occurs when restarting after stopping.

What should I do?

I'm sorry to ask for your help every time.

Thank you.

### Which car does this affect?

ev6 with hda

### Provide a route where the issue occurs

I don't know

### openpilot version

relase-c3/staging-c3/dev-c3

sunnyhaibin, 2023-08-07:
This is an upstream issue that is also affecting some HKG CAN-FD models. comma is actively fixing this issue. Once the fix is implemented in upstream, it will be merged in sunnypilot. Feel free to track the issue linked below for updates to this issue:
- https://github.com/commaai/panda/issues/1452

closed 2023-08-07 by sunnyhaibin

kyy3067, 2023-08-07:
thank you
have a nice day :)