sunnypilot/sunnypilot #201: Simplify GPS traces before uploading to openstreetmap.org

mmd-osm, 2023-08-20:
### Describe the bug

First of all, thank you for contributing GPS traces to OpenStreetMap. We'd like to bring the discussion in https://github.com/openstreetmap/operations/issues/931 to your attention. Right now, sunnypilot is the TOP 1 contributor of GPS traces according to https://planet.openstreetmap.org/statistics/data_stats.html, with more than 10 billion trace created on OpenStreetMap GPS trace database. This is putting a lot of pressure on our infrastructure: it has doubled the overall growth rate of our database, which includes much more than GPS traces.

When analyzing your traces, we have frequently seen that traces tend to be over-noded, which means, that they contain many more traces points than necessary for our purposes. We would therefore like you to consider simplifying GPS traces before uploading them to osm.org. This way, a trace with 6000 trace points could be easily reduced down to 20-30 trace points, with as little as 1m error. That's a factor of 100-300 in potential data reduction we've seen with data uploaded by sunnypilot.

We typically recommend to check out tools such as GPSBabel. Our wiki page https://wiki.openstreetmap.org/w/index.php?title=GPSBabel/Using_filters#Simplifying_tracks has some examples. 
```

gpsbabel -i gpx -f way_iv.gpx  \
    -x simplify,crosstrack,error=0.001k \
    -o gpx -F merge.gpx

```

Please let us know in case of further questions.

//CC: @tomhughes

### Provide a route where the issue occurs

N/A

### openpilot version

N/A

pnorman, 2023-09-15:
Any updates from sunnypilot on plans to simplify?

sunnyhaibin, 2023-10-24:
Thanks for bringing this to our attention. We have plans to simplify the traces in the near future.

sunnyhaibin, 2023-11-28:
Closing as the feature has been deprecated in all future releases of sunnypilot.

closed 2023-11-28 by sunnyhaibin