sunnypilot/sunnypilot #208: sunnypilot defaults to uploading fully detailed GPS traces without knowing consent

pnorman, 2023-09-15:
### Describe the bug

My understanding is sunnypilot defaults to uploading GPS traces to osm.org and allows an opt-out in the settings. These GPS traces contain vehicle make, model, year, and detailed position information that can reveal user information like home locations, work locations, where they visit, and other personal information.

The OSMF LWG is concerned that the data being uploaded is from users who have automatically been opted in to this and do not realize what they are publishing.

For automatic opt-in uploading, the traces would need to be sufficiently anonymized that personal information is not present, likely by some form of aggregation and truncation. I'm not sure what would meet those requirements.

A better option would be to default to not uploading traces, and making it opt-in. Then when a user opts in to uploading, they know that they are doing so and providing personal data.

### Provide a route where the issue occurs

N/A

### openpilot version

N/A

Stupefacient, 2023-09-17:
It doesn't default to uploading tracers to osm. You most first enable mapD which uses OSM then tracers are sent automatically. There are toggles in the settings of Sunny pilot that can disable GPS tracers and logging which are further up in interface before mapD toggle.

SomeoneElseOSM, 2023-10-14:
Is it possible to collect GPX traces and NOT upload to OSM?

mmd-osm, 2023-10-17:
@SomeoneElseOSM : is this something you could discuss with your DWG peers? It’s for sure not the first time excessive gpx uploading has been blocked for a while.

SomeoneElseOSM, 2023-10-17:
No reply here so I've asked the trace uploader to reply to the questions on the forum thread (and here).

sunnyhaibin, 2023-11-28:
Closing as the feature has been deprecated in all future releases of sunnypilot.

closed 2023-11-28 by sunnyhaibin