sunnypilot/sunnypilot #220: Recent Update gives Front Collision Error

Cx01N, 2023-10-08:
### Describe the bug

I get a warning about the front collision warning on the dev-c3 branch. I wasn’t getting this before the last few updates and don’t get it on the main branch of openpilot.
[image]

### Which car does this affect?

Hyundai Ioniq 5

### Provide a route where the issue occurs

f2a5efadd1dbf33 | 7C2023-10-07--20-20-34

### openpilot version

0.9.4

sunnyhaibin, 2023-10-10:
If you have `openpilot Longitudinal Control (Alpha)` toggle enabled, it is an expected behavior. This behavior is also the same in upstream stock openpilot. It is to communicate with users that FCW/AEB is disabled when openpilot longitudinal control (alpha) is enabled. The description of the toggle also mentions the behavior.

[image]

closed 2023-10-10 by sunnyhaibin

Cx01N, 2023-10-10:
I appreciate the heads up, I didn't realize it was intentional. Thanks!