sunnypilot/sunnypilot #244: Add Autohold and Auto High Beam support for Hyundai Sonata

Sliicy, 2023-11-24:
Hello!
I was wondering if you could add 2 toggles in settings for the Hyundai Sonata 2021 (possibly other variants too) that if enabled, broadcast a CAN message once on engine start to make the car believe that Autohold or Auto High Beam has been pressed by the driver?

Reason: It would be really awesome if the car could remember Auto High Beam's last state as well as Autohold, so that the driver doesn't have to turn those 2 buttons each time if they like using it.
In Toyota cars at least, the Auto High Beam setting is remembered when the engine is started each time.

Thank you!

Sliicy, 2023-11-24:
Here are 2 threads where others discuss the issue of the car not remembering the setting each time:

https://www.hyundai-forums.com/threads/auto-hold-turns-off-after-car-restarts.546513/

https://www.hyundai-forums.com/threads/smart-high-beam.330961/

sunnyhaibin, 2023-11-28:
Thanks for the feature request. Closing as this is more of a `feature request` than an `issue`. Please have it reposted in the `#feature-requests` section in the [sunnypilot discord server](https://discord.sunnypilot.com). We will follow up in the new thread in the sunnypilot discord server once it is reposted. Thank you.

closed 2023-11-28 by sunnyhaibin