sunnypilot/sunnypilot #245: Zig-zag motion on Hyundai Ioniq

poppyseedDev, 2023-11-26:
### Describe the bug

### Description:
**Overview:**
We have observed a consistent zig-zag motion, commonly referred to as 'lane hunting' or 'oscillation', in the Hyundai Ioniq when operated under autopilot. This issue is particularly noticeable on highways and straight roads, where the vehicle should ideally maintain a steady course within the lane.

**Details:**
- **Conditions:** All, more pronounced when there is strong wind.
- **Frequency:** The oscillation occurs consistently and seems to be a systematic issue rather than an occasional anomaly.

**Behavior:**
- The vehicle tends to overcorrect its steering when trying to align with the center of the lane.
- This results in a zig-zag pattern of movement within the lane, causing discomfort for passengers and potentially confusing other drivers.

**Expected Behavior:**
- The vehicle should maintain a smooth, steady course within the lane without excessive steering adjustments.

### Action Requested:
- Investigate the underlying cause of this behavior.
- If there are any parameters or settings within the autopilot system that users can manually adjust to mitigate this issue, detailed instructions or insights would be greatly appreciated.
- In case the solution involves adjusting code within the autopilot system, we would be grateful for guidance or recommendations on specific changes that could be implemented to improve performance.

### Which car does this affect?

Hyundai Ioniq 2021 

### Provide a route where the issue occurs

Slovenia's highway

### openpilot version

latest

poppyseedDev, 2023-11-26:
https://github.com/sunnyhaibin/sunnypilot/assets/30662672/0be9e8b0-08cd-45f1-9548-969195f83802

sunnyhaibin, 2023-11-28:
This is a `tuning` issue that could be caused by the model's output and/or `torqued` live values. We have numerous discussions in the [sunnypilot discord server's](https://discord.sunnypilot.com) `#tuning` and `#tuning-nnff` channels  on this topic and how to mitigate this.

closed 2023-11-28 by sunnyhaibin