sunnypilot/sunnypilot #247: OpenPilot Unavailable paramsd Temporary Error

tmichutka, 2023-12-09:
### Describe the bug

I’m getting the error “OpenPilot Unavailable paramsd Temporary Error” on my 2018 Toyota Prius Prime with Comma 3x running SunnyPilot 0.9.5.2, SDSU, and ZSS.  I’ve only found one other similar issue on GitHub, and it was on a 2017 Toyota Highlander (OpenPilot Issue #30598).  However, unlike that user I don’t have a steering wheel spacer.

This issue is a bit scary because when this “paramsd temporary error” happens, OpenPilot typically steers sharply/dangerously either left or right.  I have to immediately grab the wheel and disable ACC.  Sometimes if I try 5 or 10 times SunnyPilot will activate.  Sometimes it works the first time.  I’ve tried reinstalling SunnyPilot and resetting all settings to default, neither had any effect on the issue.  

### Provide a route where the issue occurs

77c43558942b8272|2023-12-09--10-56-13

### openpilot version

0.9.5.1

sunnyhaibin, 2023-12-09:
Try installing the latest `dev-c3` and see if the issue is resolved, it is confirmed to be fixed in `dev-c3`. If there are any further issues feel free to reach out to us in the [sunnypilot Discord server](https://discord.gg/sunnypilot).

closed 2023-12-09 by sunnyhaibin

tmichutka, 2023-12-09:
Thank you so much, @sunnyhaibin!  Both for your amazing fork and the lightning fast reply!  Installing dev-c3 on my C3X as I type.  If I encounter any further issues I'll reach out on Discord.  Thanks again!  =)