sunnypilot/sunnypilot #253: LKAS Fault

tonnytsang96, 2023-12-17:
### Describe the bug

When I turn the wheel left or right maximum. 
COMMA 3 show LKAS fault. Thank you.

Car: Toyota Corolla 2020yr

[image]

### Provide a route where the issue occurs

60dcd78f23ffcb45|2023-12-16--21-09-52

### openpilot version

0.9.5.2

altesean, 2023-12-19:
This has also affecting me on a 2020 Rav4 a few times, seems to happen when at full steering wheel lock during U-Turns with force applied to keep it at max during the turn if you hold the wheel loose just before lock it doesn't trip.

siddchou, 2024-01-13:
I am also getting this error in 2024 cx-5 
[image]

sunnyhaibin, 2024-06-14:
This should be resolved in the latest branches. If it is still an issue, please feel free to open another issue.

closed 2024-06-14 by sunnyhaibin