sunnypilot/sunnypilot #279: Some Hondas Have Noisy Gas Pedals

joeljacobs, 2024-02-08:
### Describe the bug

All 2018-2022 model Odysseys and some Acuras have noisy gas pedals. This means that when using the Comma Pedal, the gas pedal registers false presses on the gas just due to how high the noise floor is. 
In the past, I was able to fix this for my Odysseys and the Acura rdx by changing the following lines in opendbc/honda_odyssey_exl_2018_generated.dbc:
 SG_ INTERCEPTOR_GAS : 7|16@0+ (0.253984064,-120.3) [0|1] "" EON
 SG_ INTERCEPTOR_GAS2 : 23|16@0+ (0.126992032,-120.3) [0|1] "" EON

And I had to change the line in panda/board/safety/safety_honda.h:
const int HONDA_GAS_INTERCEPTOR_THRESHOLD = 496.8;

This always fixed the problem before. But since they changed the architecture of OpenPilot, changing the ...generated.dbc file does nothing - probably since it now defaults to (1,0) which is different than it used to be. 

I would just do a pull-request, but I don't know how to do it now. I'm hoping you do. 

### Which car does this affect?

Honda Odyssey and Acura rdx

### Provide a route where the issue occurs

6993681710231e24%7C2024-02-07--18-03-10

### openpilot version

0.9.5 but any after 0.7

joeljacobs, 2024-06-29:
I particularly need you and the community's help now that the pedal is no longer a priority to Comma. Please help!

Stupefacient, 2024-08-27:
Is this issue still a thing? May I close it @sunnyhaibin @joeljacobs?

sunnyhaibin, 2025-03-16:
@aragon7777 you still working on this?