sunnypilot/sunnypilot #299: Update Check Requirements

ajowens001, 2024-03-08:
Hi there!

Thank you for your hard work on this!

I'm writing because I recently updated my device, and now I notice that there is an update check that must successfully reach the Internet in order to allow the software to start up.

This doesn't seem all that important to me and I am requesting this pop-up be removed and to have this new update requirement nullified. 

While I understand that all we need to do to dismiss the notification is to tap a couple buttons, but this new requirement takes a large piece of quality of life functionality out of our devices.  

I'm sure I'm not the only one to enjoy having our device boot up and be ready to drive without extra user input.  I believe the unit was designed with this notion in mind and would like that option back. 

[image]

Thank you for your effort, time and consideration.

ajowens001, 2024-04-10:
Bump.  Anyone able to give this some attention?

sunnyhaibin, 2024-06-14:
This may happen due to some issues with time synchronization. Try to factory reset and reinstall the branch. If the issue still persists, please feel free to open another issue.

closed 2024-06-14 by sunnyhaibin