sunnypilot/sunnypilot #303: 2022 Honda Civic using 3x

matluvhonda, 2024-03-12:
### Describe the bug

Fully supported vehicle here (2022 Honda Civic). After installing openpilot, the comma 3x went to dashcam mode because of "unsupported vehicle". Installed sunnypilot and forced the vehicle, now I have a CAN error (checked all connections, reset the 3x, even unplugged from lkas to ensure nothing was wrong with car hardware). Sunnypilot seems to be able to take lateral (steering) control of the vehicle, but not longitudinal (gas/brake) control. The 3x also can't seem to connect to GPS and my SIM card doesn't seem to work in it. 

I'm kind of a newb, but I do have some python experience and am willing to help get this sorted before the 30 day return period is up with comma for returning the 3x. Basically, I don't want a $1500 dash cam. 

My car is a 2022 Honda Civic Si (I'm wondering if part of the issue is with my trim level and having a manual transmission). 

As for the GPS and SIM issues, I'm wondering if I have a defective 3x? Comma's support has been nothing but automated emails so far. Hoping sunnypilot team is more helpful. 

### Which car does this affect?

2022 Honda Civic

### Provide a route where the issue occurs

21e3cd235b4e85c2/2024-03-07--18-50-48/0

### openpilot version

0.9.6?

### Additional info

Not sure where to find the version of openpilot/sunnypilot. The device was setup last week. Pretty sure it's 0.9.6 or 0.9.6.1

When attempting to upload a route, the dashcam doesn't seem to be uploading my routes either. Only 3 "drives" are greater than 1 minute (with probably a dozen drives so far). The 3x is connected to my home WiFi and has a new T-Mobile phone SIM with unlimited data in it.

w2913212835, 2024-03-26:
I'm experiencing almost the scenario like you. Mine's RAV4 2022. Here's what i do to save myself from the situation.

First, factory reset the comma 3x, then, start the installation from the beginning. software download URL type this: bit.ly/sp-release-c3   (https:// is not necessary for comma 3x) 

sunnypilot should support manual transmission car. As for GPS and sim problem, i wondered if the system has dirty data, that might cause some problem.

if you do the things above, try to switch the usb cable, or change the L shape side to harness connector and the straight side to comma 3x

YI418LT, 2024-04-26:
同じ事象が2020 CIVIC(SHHFK8)でも起こります。プログラムはCIVIC hatchbackと同じはずですが、もし何か改善策がわかる人が居ましたら教えて下さい。

sunnyhaibin, 2024-06-14:
Manual transmissions cars are not officially supported in openpilot, and sunnypilot currently does not support Honda models iwth manual transmission as such change has never been planned. To support manual transmissions variants of Honda models, a car port is required.

Feel free to reach out to the community on Discord regarding adding support for manual transmission for Honda models. https://discord.gg/sunnypilot

closed 2024-06-14 by sunnyhaibin