sunnypilot/sunnypilot #324: [Question] The master branch supports being build?

ebellumat, 2024-05-04:


ebellumat, 2024-05-05:
I'm getting this error when I try to build the master branch.
```bash

./launch_openpilot.sh 
/home/ebellumat/Documents/GitHub/sunnypilot/launch_chffrplus.sh: line 7: /data/openpilot/launch_env.sh: No such file or directory
ln: failed to create symbolic link '/data/pythonpath': No such file or directory
/home/ebellumat/Documents/GitHub/sunnypilot/launch_chffrplus.sh: line 82: tmux: command not found
selfdrive/ui/translations/main_ar.ts: needs update
selfdrive/ui/translations/main_de.ts: needs update
selfdrive/ui/translations/main_fr.ts: needs update
selfdrive/ui/translations/main_ja.ts: needs update
selfdrive/ui/translations/main_ko.ts: needs update
selfdrive/ui/translations/main_pt-BR.ts: needs update
selfdrive/ui/translations/main_th.ts: needs update
selfdrive/ui/translations/main_tr.ts: needs update
selfdrive/ui/translations/main_zh-CHS.ts: needs update
selfdrive/ui/translations/main_zh-CHT.ts: needs update
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.
system/loggerd/bootlog.cc: bootlog to /home/ebellumat/.comma/media/0/realdata/boot/00000005--2f9e54226e
missing public key: /home/ebellumat/.comma/persist/comma/id_rsa.pub
selfdrive/ui/translations/main_ar.ts: needs update
selfdrive/ui/translations/main_de.ts: needs update
selfdrive/ui/translations/main_fr.ts: needs update
selfdrive/ui/translations/main_ja.ts: needs update
selfdrive/ui/translations/main_ko.ts: needs update
selfdrive/ui/translations/main_pt-BR.ts: needs update
selfdrive/ui/translations/main_th.ts: needs update
selfdrive/ui/translations/main_tr.ts: needs update
selfdrive/ui/translations/main_zh-CHS.ts: needs update
selfdrive/ui/translations/main_zh-CHT.ts: needs update
Manager failed to start
Traceback (most recent call last):
  File "/home/ebellumat/Documents/GitHub/sunnypilot/selfdrive/manager/./manager.py", line 270, in <module>
    main()
  File "/home/ebellumat/Documents/GitHub/sunnypilot/selfdrive/manager/./manager.py", line 239, in main
    manager_init()
  File "/home/ebellumat/Documents/GitHub/sunnypilot/selfdrive/manager/./manager.py", line 161, in manager_init
    p.prepare()
  File "/home/ebellumat/Documents/GitHub/sunnypilot/openpilot/selfdrive/manager/process.py", line 213, in prepare
    importlib.import_module(self.module)
  File "/home/ebellumat/.pyenv/versions/3.11.4/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/ebellumat/Documents/GitHub/sunnypilot/system/fleetmanager/fleet_manager.py", line 5, in <module>
    from flask import Flask, render_template, Response, request, send_from_directory, session, redirect, url_for
ModuleNotFoundError: No module named 'flask'`
![Screenshot from 2024-05-05 12-40-46](https://github.com/sunnypilot/sunnypilot/assets/26942680/e7ab572e-e437-4399-8753-d025633ce450)

ebellumat, 2024-05-05:
Issue solved commenting the FleetManager, but @sunnyhaibin Why the tools folder are available only on the master branch? I'm curious.

rav4kumar, 2024-06-22:
`sudo pip install flask`

The master branch is a sync branch that we sync from comma/master. The rest are prebuilt (precompiled binary). Whatever you do not see on sunnypilot/master is not meant to be public.

Thank you,

closed 2024-06-22 by rav4kumar