sunnypilot/sunnypilot #345: ALC disengage not working with either LKAS button nor breaking pause toggle on 2024 RAV4 Limited

akutruff, 2024-06-13:
### Describe the bug

Title.  Hitting the LKAS button does not disengage the steering.  The toggle for pause ALC on break is not working either.

This is a rather scary and unsafe one.  The only thing that will disable steering control is turning off cruise completely.

### Which car does this affect?

2024 RAV4 Limited 

### Provide a route where the issue occurs

Not available. No instructions provided.

### openpilot version

0.9.6.1

sunnyhaibin, 2024-06-13:
Thank you for reporting. Here are some important points to consider:

> Hitting the LKAS button does not disengage the steering

To use this feature, you must enable the appropriate toggle `Allow M.A.D.S. togglign w/ LKAS Button (Beta)`. Please note that the LKAS button on most Toyota vehicles has inconsistent activations with the button press with sunnypilot/openpilot. Hence, having this activation behind a toggle is expected behavior at its current stage.
[image]

> The toggle for pause ALC on break is not working either

The car must be in the offroad state to enable or disable this toggle. Starting from sunnypilot version 0.9.7.1, you can use the Force Offroad feature to trick the device into the offroad state to interact with this toggle.
[image]

> This is a rather scary and unsafe one

Please remember that sunnypilot is a Level 2 ADAS. This means it requires the driver's full attention and engagement at all times. If you believe the system is dangerous, please understand its limitations and avoid using it if you feel it's unsafe. The current universal way to disable steering control in Toyota cars is to turn off the cruise completely by pressing the CRUISE MAIN button.

closed 2024-06-13 by sunnyhaibin

akutruff, 2024-06-14:
Are these exceptions for Toyota documented anywhere else? The fact that it was a surprise on my first drive is why it was unsafe.  I read the docs and had no way of knowing my car would behave to the contrary.

The UI toggle was useable and I was able to select "pause steering" even though I was not in offroad mode.

sunnyhaibin, 2024-06-14:
> Are these exceptions for Toyota documented anywhere else? The fact that it was a surprise on my first drive is why it was unsafe. I read the docs and had no way of knowing my car would behave to the [...]

> The UI toggle was useable and I was able to select "pause steering" even though I was not in offroad mode.

The toggle is designed to be non-toggleable when the car is on (or the device is in `onroad` mode) since changing this toggle requires the car to be power cycled. If you were able to interact with it when the car was in `onroad` mode, make sure to power cycle the car to apply the change.

We are working on updating the documentation to reflect certain limitations with specific platforms, so some areas may not yet reflect the latest changes. Regardless of the `LKAS` button behavior with Toyota/Lexus, the following from the current documentation still applies:
- `CRUISE (MAIN)` button disables sunnypilot completely when `OFF` (strictly enforced in panda safety code)

If you believe certain features are unsafe, please remember that sunnypilot is a Level 2 ADAS, and the driver must always pay attention and be ready to take over should any issues or unexpected behaviors arise. In the meantime, pressing the `CRUISE (MAIN)` button will disengage all controls from sunnypilot.