sunnypilot/sunnypilot #360: When longitudinal control is turned on, an AEB warning occurs.

itkairos2023, 2024-07-21:
hello? When longitudinal control is turned on, an AEB warning occurs.

My vehicle is a Staria vehicle.

Staria uses camera SCC, and SCC data comes out from the camera.

This is a rare vehicle that can be developed without AEB warnings.

For safety reasons, is it possible to modify the AEB warning (so that the AEB function is not disabled) even if Long Con is enabled?

Other branches, for example Carrot Pilot, have been modified so that AEB warnings do not appear.

https://github.com/ajouatom/openpilot/tree/carrot-master

devtekve, 2024-08-02:
@itkairos2023 I believe this needs to be updated upstream first. We can do it if you can give us a route where you have had FCA and / or AEB so we can confirm that it works in your car before we can leave the AEB warning off. 

For context, the AEB warning is forced active because traditionally we disable the radar which in turns disables AEB. So we notify the user that AEB is off. If we can have confirmation the car is indeed retaining AEB then we can stop turning that warning on.

Stupefacient, 2024-08-27:
Stock openpilot problem. As @devtekve stated comma needs to fix this not us. If we were to fix it ourselves it could result in the fork getting banned for violating, safety policy.

closed 2024-08-27 by Stupefacient