sunnypilot/sunnypilot #412: Automatic Windshield Wipers Toyota Corolla Hybrid ($150 reward)

qf0, 2024-08-15:
Using the Comma 3x camera, I would like a feature to automatically turn on my cars (Toyota Corolla Hybrid 22) windshield wipers via can messages. I'm willing to pay $150 as a reward.

RisaKirisu, 2024-08-23:
Unfortunately, vision based rain detection is hard and very unreliable. 

For example, Tesla tries to do rain sensing with pure vision and it's the most unreliable part of the car - absolutely abysmal user experience even after years of updates. Tesla has 8 cameras all around the car btw.

Part of the difficulty is that the cameras used for self-driving can barely see rain - they need to focus at very far distance in order to drive, and rain droplet on the windshield in front of the cameras are only centimeters away. It's also hard for the cameras to capture the falling rain - just try to take a picture of rain fall and see if you can capture the falling rain clearly.

Stupefacient, 2024-08-27:
Unfortunately this is not in the scope of Sunnypilot and we have no plans to add it as this is driving software not object or weather software detection software. Also we do have the right cans signals to  override/control the whatever blades.

closed 2024-08-27 by Stupefacient