sunnypilot/sunnypilot #643: Model Manager does not check file integrity or existence of metadata

devtekve, 2025-03-02:
I noticed that if a model's metadata is deleted, the model fails to load, but the issue is not immediately clear.

When I originally implemented this, I intended for it to perform integrity checks. Specifically, if the selected model had a different hash than the remote version or if the local file was missing, the manager would automatically download it. Ideally, it should also clean up the `ModelManager_ActiveBundle` parameter in such cases.

The relevant code is located in `sunnypilot/models/manager.py`

devtekve, 2025-03-02:
@sunnyhaibin FYI, unrelated to the current work streams we have but something we'd need to work on for more robustness

sunnyhaibin, 2026-02-06:
Raylib rework completed.

closed 2026-02-06 by sunnyhaibin