sunnypilot/sunnypilot #703: camera malfunction roadcamera, wideroadcamera

life982001, 2025-03-23:
### Describe the bug

Describe the bugs

Upgrade to Sunnypilot dev-c3-new,
and the screen displayed "camera malfunction roadcamera, wideroadcamera." 
when release-c3 or dev-c3 is OK.

C3X info:
The Dongle ID : 647c045f0ff7f1fe
The serial : 1a42809d

Car info:
Mazda CX5 2023 ( Taiwan , made in Japan )

### Provide a route where the issue occurs

647c045f0ff7f1fe/00000009--6dfd1c1252

### openpilot version

0.9.8

### Additional info

route : https://connect.comma.ai/647c045f0ff7f1fe/00000009--6dfd1c1252

devtekve, 2025-03-23:
Related: 
* https://github.com/commaai/openpilot/issues/34903
* https://github.com/commaai/openpilot/pull/34611

We are open to PRs from the vendors to fix this as long as it's reasonable and has zero impact on other hardware.

devtekve, 2025-03-23:
@life982001 looks like what I said earlier does not apply to you. Got the wires crossed for a second. When did you experience this? what's the build number ?

life982001, 2025-03-23:
@devtekve 

About build , I found the infos from route:

'deviceType': 'tizi',
              'dirty': True,
              'dongleId': '647c045f0ff7f1fe',
              'gitBranch': 'dev-c3-new',
              'gitCommit': '09f1865576ed9839e3e2eb5e9cd138d70656d79f',
              'gitCommitDate': "'1742460245 2025-03-20 08:44:05 +0000'",
              'gitRemote': 'https://github.com/sunnyhaibin/openpilot.git',
              'gitSrcCommit': '',
              'gitSrcCommitDate': '',

If there is any other information needed, please let me know. Thank you!

life982001, 2025-03-23:
Additional information:
I also tried installing a forked version from another car enthusiast, which is also 0.9.8. The author mentioned that he removed a camera encoding, and after installing it, the issue was resolved! Just letting you know.

devtekve, 2025-03-25:
thanks @life982001 ! have you created an issue report on upstream OP ? this is certainly an upstream issue specially with an official device. I will see if we can do something on our end in the meantime a better fix is made

devtekve, 2025-03-25:
@life982001 do you mind linking the author of the fix here so I can take a look at what exactly he did? I myself am not familiar with the hardware / fw programming so the more info the better

devtekve, 2025-03-25:
# Investigation Summary

1. **Initial Findings:**  
   Based on [@royjr's comment](https://github.com/commaai/openpilot/pull/34701#issuecomment-2680294881) and similar error logs reported by @life982001, it appears both were encountering the same issue.

2. **Issue Tracking:**  
   An issue was opened upstream ([Issue #34663](https://github.com/commaai/openpilot/issues/34663)) and linked to @royjr's comment by @deanlee ([comment link](https://github.com/commaai/openpilot/pull/34701#issuecomment-2680340549)).

3. **Resolution on Upstream:**  
   The upstream issue was closed via [PR #34758](https://github.com/commaai/openpilot/pull/34758). Notably, @royjr (an SP developer) has not reported further issues.

4. **Persisting Issue on `dev-c3-new`:**  
   @life982001 experienced the issue on the `dev-c3-new` prebuilt **1312** (from [this commit](https://github.com/sunnypilot/sunnypilot/tree/09f1865576ed9839e3e2eb5e9cd138d70656d79f)), built from [this branch](https://github.com/sunnypilot/sunnypilot/tree/d13b6345aca157ea9af56dda8f3364e634a4c0df). Despite the fix from PR #34758 being deployed, the issue persists.

5. **Additional Observation:**  
   A separate fork, which removed camera encoding (likely referring to changes from [PR #34701](https://github.com/commaai/openpilot/pull/34701)), reportedly resolved the issue.

devtekve, 2025-03-25:
@life982001 would you mind trying `test-revet-encoding-changes-prebuilt` and seeing if it works for you? (_**note** the typo on `revet`_) 
if it does, would you mind trying `staging-c3-new` again just to double check ?

life982001, 2025-03-25:
@devtekve 
Sure, I'm happy to help with the testing.  
I'll first test "test-revet-encoding-changes-prebuilt", and if it works, I'll then test "staging-c3-new".

life982001, 2025-03-25:
@devtekve 
The test is done , after installed "test-revet-encoding-changes-prebuilt ,but the result is still the same.

life982001, 2025-03-25:
@devtekve 

Here’s another version with the author's modifications for your reference.

[image]

devtekve, 2025-03-25:
Thanks @life982001 so the issue is not the encoding then, I will bring that back and comment the the registers there

life982001, 2025-03-25:
@devtekve 
Great!
This version can boot up normally!

[image]

devtekve, 2025-03-25:
@life982001  could you also try `https://github.com/sunnypilot/sunnypilot/tree/test-revet-encoding-changes-toggle-params-prebuilt` but flipping the toggle under developer panel ?

life982001, 2025-03-26:
@devtekve 
This build is ok too,
But I dont understand what it means "flipping the toggle under developer panel"?

[image]

life982001, 2025-03-26:
@devtekve 

If you mean "use legacy camera sensor.....",
No matter which one I switch to, everything is normal, and the previous error message does not appear.