sunnypilot/sunnypilot #743: Longitudinal: Set Speed Increment Adjustment

sunnyhaibin, 2025-04-01:


devtekve, 2025-04-10:
Here we want to support changing the amount of increment and also inverting it

sunnyhaibin, 2025-04-10:
https://github.com/sunnypilot/sunnypilot/commit/6ad79d3e81239fb2d8864d42c95f5e1b06e456a2

sunnyhaibin, 2025-04-10:
- Setup a method and explicitly gate a few brands in `root/sunnypilot/car/interfaces.py`
- Gate these brands for now until we understand them a bit more and/or there's a clear path for these:
  - Tesla
  - Rivian
  - Ford

devtekve, 2025-04-10:
> Yes, it is working just as I'd expect it to. (But I'm out of the habit again of pressing long for unit changes and short for blocks of 5). My name suggestion? Swap ACC speed changes.

Via https://discord.com/channels/880416502577266699/994915271213064222/1359980265413283940

closed 2025-09-09 by devtekve