sunnypilot/sunnypilot #828: Allow starting in always offroad mode

devtekve, 2025-04-15:


wtogami, 2025-04-15:
The other reason is various EV modes like Toyota "Room Mode" is seen by OP as "Ignition On" but the car can't move. I use that as an air conditioned office sometimes. I don't want the camera to record me and my family during those times, and the route recorded during that time is useless.

Can a button to turn off Offroad mode be added to the main screen? Then you don't need to open the settings and scroll down to turn SP back on.

nayan8teen, 2025-04-21:
Is this already being worked on? If not, I could take a stab at it.

devtekve, 2025-04-21:
> [@devtekve](https://github.com/devtekve) Is this already being worked on? If not, I could take a stab at it.

Go for it ! We haven’t been working on it yet.

I had in mind 2 things: properly detecting cars that have that sort of “camping” mode, and maybe having the toggle to start always forced off-road and when that’s active, show on the home page a button to go onroad (similar to where the firehouse button is or where the navigation destination used to be).

I would not add anything in the onroad screen, so no button to go off-road from the onroad page. You should only ever be able to go off-road from the settings and when the control is not active

wtogami, 2025-04-22:
Idea 4. Route recording geofencing

Define an area. If you are within that area it stops recording.*
The goal here is privacy to be able to share routes publicly without showing exactly where you live.

* Instead of stop recording entirely, what if it is recorded to a separate local store that isn't uploaded to Connect? That way if you need that portion of the recording it is available in a different way.

nayan8teen, 2025-04-28:
Started playing with this. When we say "start in offroad mode", should that be only at boot? Or should we force always offroad mode every time the device transitions from onroad->offroad so that it will NEVER go into onroad state without the user explicitly exiting always offroad.

wtogami, 2025-04-28:
"start in offroad mode" at boot.

That would be the simplest workaround for the above variety of other problems.

nayan8teen, 2025-04-28:
> "start in offroad mode" at boot.
> 
> That would be the simplest workaround for the above variety of other problems.
> 
> 

Yeah, that's what I started with. Just needs some cleanup before I submit a PR for it - probably later tonight.

nayan8teen, 2025-04-29:
PR Created: https://github.com/sunnypilot/sunnypilot/pull/861

nayan8teen, 2025-04-30:
For adding an Exit button on Offroad screen.. preference between these 2?

[image]

[image]

wtogami, 2025-04-30:
I like both of them.

wtogami, 2025-04-30:
BTW does anyone else think "Always Offroad" is confusing as a name?
It will soon be more confusing with the option to boot into Always Offroad.