sunnypilot/sunnypilot #862: DevTekVe’s pending queue

devtekve, 2025-04-30:
Please write down on the comments if you have anything pending from mee , the link to the PR and a summary of what you expect

Discountchubbs, 2025-04-30:
Review model selector pr and add to gitlab json 
Ref commits and shas are pinned to discord.

nayan8teen, 2025-04-30:
1. Review [Start in Always Offroad] (https://github.com/sunnypilot/sunnypilot/pull/861) and move to dev if looks good.
2. Review my attempt at using cereal for acc increments on my fork: [acc-custom-cereal](https://github.com/nayan8teen/openpilot/commits/acc-speed-increment-cereal/) - I haven't merged it into the existing PR yet since I didn't want to break something that has already been working in dev & people seem to be actively using it.

nayan8teen, 2025-05-01:
> > 1. Review [Start in Always Offroad](https://github.com/sunnypilot/sunnypilot/pull/861) and move to dev if looks good.
> > 
> >     2. Review my attempt at using cereal for acc increments on my fork: [acc-custom-cereal](https://github.com/nayan8teen/openpilot/commits/acc-speed-increment-cereal/) - I haven't merged it into the existing PR yet since I didn't want to break something that has already been working in dev & people seem to be actively using it.
> 
> for the second point, I've made this draft PR to visualize it https://github.com/sunnypilot/sunnypilot/pull/865/files

The changes do work as expected.. tested onroad as well.. Just want to make sure the way I've implemented it using cereal makes sense.

closed 2025-07-16 by Discountchubbs