sunnypilot/sunnypilot #913: (HKG Angle LFA2) Slow-to-Center After Sharp Low-Speed Turns

devtekve, 2025-05-11:
**Focus:**
At very low speeds and high angles, the steering system retains lock due to rate limits on actuator unwinding, preventing expected self-centering behavior after a sharp turn.

**Symptoms:**

* During a sharp low-speed turn (e.g., 90° at \~9 mph), the wheel holds the turn longer than expected.
* Driver has to manually counter-steer to unwind.

**Potential Brainstorm Ideas:**

* Adjust rate limits to allow faster return-to-center after a sharp low-speed turn.
* Use speed-dependent rate limits to relax constraints where safe.
* Detect end-of-turn conditions and re-engage centering logic sooner.

devtekve, 2025-05-11:
I had an idea to address this issue: reduce the torque almost completely when the requested steering angle is in the direction of unwinding, and the delta is too large. The goal would be to allow the steering wheel to turn more quickly in response to external factors like road friction.

However, this approach could potentially negatively affect [Issue #912](https://github.com/sunnypilot/sunnypilot/issues/912), since, during normal cruise, we want higher torque to help maintain the center position.

To strike a balance, we could implement a 'torque curve' where, the further we are from the target angle (especially when the angle is in the unwind direction), the torque gradually weakens. As we get closer to the target angle, the torque would ramp back up. This way, the farther we are from the target (in the unwind direction), the weaker the torque, and conversely, the closer we are, the stronger the torque.

TheSecurityDev, 2025-05-14:
For what it's worth, I've noticed the same thing on 2021 Toyota Camry Hybrid after the latest models.